jpconstantineau/ErgoTravel #11: Complete ordering guide

jpconstantineau, 2018-03-25:


dbordak, 2018-06-21:
I'm interested in buying an ergotravel, but the current ordering instructions are a bit insufficient for me. If you could help me through that, I could document the process and PR it. I know it'd probably be just as much effort on your part, but I find talking easier than writing, personally :)

jpconstantineau, 2018-06-21:
I understand that the ordering instructions are a bit scarce right now. I have a Group Buy running right now.  If you are willing to wait and join in, you will be able to get it at a better price than ordering small quantities from the PCB fabs.  Once the GB is complete, I will have a complete list of parts and where to obtain them.

Also, I am making small changes to the gerber files as I am still getting some input from the beta testers.

I do have some beta tester boards available if you are really in a rush to get things and want to contribute to testing the board and building process. See [here](https://www.reddit.com/r/mechmarket/comments/8ssgu4/gb_ergotravel_group_buy_update_2/) for the GB details.

If you do want to contribute to the instructions and make the necessary changes, you can start from the instructions [here](http://www.40percent.club/2017/03/ordering-pcb.html).  They are not completely up to date but very close.

dbordak, 2018-06-22:
I actually already signed up for the GB, but I'm also kinda hype for it and want one now :P 
It fits all my needs (or at least, what I think I've determined I want) for a keyboard without anything extra -- and I've found that forcing myself to use chording rather than reaching for an extra row/column feels better.

jpconstantineau, 2018-10-14:
I have updated the ordering guide; however, I am leaving is as a BOM with a few links. Closing the issue

closed 2018-10-14 by jpconstantineau