jpconstantineau/ErgoTravel #14: Holes for PCB mounted switches are too tight

ckofy, 2018-03-27:
I’ve tried to insert the Zealios switch to the PCB and it is very hard going. I’ve measured the diameter of the switch’s pins as 1.7 mm.

jpconstantineau, 2018-03-27:
Cherry Specs indicate putting holes at 1.7mm diameter.  Keebio uses 1.8mm holes for his footprints.

ckofy, 2018-03-27:
I've found what causing that issue. Pins of Zealios switches have a little excess of of plastic on the inner sides making them not perfectly round. Grinding that excess out with a little file makes the switch to fit nice in the ErgoTravel PCB, but after that procedure it starts to fall out of the Iris PCB.

jpconstantineau, 2018-03-27:
Production Tolerances on drill sizes could also be a factor:
From JLCPCB Capabilities page:
Hole Size Tolerance | ±0.08mm | e.g. For the 0.6mm hole, the finished hole size between 0.52mm to 0.68mm is acceptable.

jpconstantineau, 2018-11-15:
I won't be making the changes to the footprint.

closed 2018-11-15 by jpconstantineau