jpconstantineau/ErgoTravel #27: Move ErgoTravel to use split_common library (QMK)

SethBarberee, 2018-07-26:
A lot of the files here are repetitive and from glancing the code, the keyboard could use the split_common library in QMK. Recently did a PR for the Iris for this exact thing, however I can't verify the PR would work until my ErgoTravel comes in whenever from the GB. If you can't get around to it, I can work on it when I get my ErgoTravel in from the GB.

Link to PR for reference: https://github.com/qmk/qmk_firmware/pull/3458

jpconstantineau, 2018-07-26:
Hi Seth,  The ErgoTravel is already in QMK but as you indicate, it doesn't use the split_common library.  Since I based the original ErgoTravel code on the Iris, I suspect it will be possible to use this library too.

SethBarberee, 2018-10-27:
Closing since one of the main QMK collaborators has ported it to the split_common code

closed 2018-10-27 by SethBarberee

jpconstantineau, 2018-10-27:
Thanks Seth for the update.