jpconstantineau/ErgoTravel #28: Serial implementation

Derecho, 2018-08-22:
I was looking at your code as a friend is making their own ErgoTravel keyboard, and something's puzzling me a little.

So as far as I understand, to link the two halves of the keyboard, either I2C or "regular" serial is used (with a TRRS cable). I see the defines in the code to choose between the two protocols, but the files for serial seem to be missing in this repository? I did find them in this merge commit: https://github.com/qmk/qmk_firmware/pull/3129/files

Now I might be misinterpreting things, but from the serial.c and serial.h I can see there, it looks like software serial is used (bitbanging) on port PD0. Why? I'd assume hardware serial (on PD2/PD3) would be preferred, since that makes timing a lot easier and it saves you some instructions. I wouldn't be surprised if you could push that to greater speeds too, thus reducing latency (on the slave half).

One reason I can think of, but this is just speculation, is that I2C was implemented first (of which the pins are on PD0/PD1) and serial support was added later, and that one wanted to keep pin compatibility between the two. Is this it, is there another reason, or did I completely misinterpret the serial operation?

jpconstantineau, 2018-08-22:
The implementation is derived from the Iris and ultimately the original split board: the let's split.  The Let's Split uses serial big-banging or I2C on the same lines.  In order for everything to be consistent with everyone's design, I kept the same lines.

I don't know if tx/rx was ever tried/used between two halves. I agree with you that it would make sense.  However, many implementations use bi-directional serial over a single line so that the extra wire of the TRRS line is used for the data line of the RGB underflow strips.  I suspect that this is the primary reason why a single line is used.  

There has been a few attempts to get RGB to work over I2C but I am not aware of any that have been completely successful. (Getting the RGB patterns to sync between the halves isn't trivial. With the extra line connecting DO to DI of the master/slave strips, one doesn't have to worry about syncing the two sides)

SethBarberee, 2018-08-30:
Actually, the [Let's Split Eh?](https://github.com/qmk/qmk_firmware/tree/master/keyboards/lets_split_eh) is purely I2C and has support for RGB. The support of the Let's Split Eh? of RGB started the split common library of QMK mentioned in #27. Currently, only static RGB is supported. Animations haven't been implemented yet by Canadian (the main guy behind Let's Split Eh?).

closed 2021-11-24 by jpconstantineau