jpconstantineau/ErgoTravel #29: PCB Choc tested?

Cwejman, 2018-10-17:
Hi,

I'm eager to order and assemble my first split keyboard. I've ended up here after trying to make up my mind for a while. I was about to attempt to port the layout to choc switches but it would seem a fresh commit's been made. Is this a finished layout and has it been tested?

Most kind of you to share what you create! Please share if any help is needed.

/ Jona

jpconstantineau, 2018-10-17:
No. It's not been produced or tested yet.  There are already some
modifications I need to make to the PCB.

jpconstantineau, 2018-10-25:
@Cwejman if you are interested in the chocolate switches layout, what features would you also be interested in? I am considering adding smd LEDs (either rgb or not).

mattkelly, 2019-03-03:
@jpconstantineau I'm also interested in trying out the choc layout. Any hints as to what PCB modifications are already known to be needed? I'm not interested in LEDs or other features (well, maybe Bluetooth at some point) - mostly just the switches themselves for a lower profile.

jpconstantineau, 2019-03-03:
I do have prototype PCBs on hand and am looking to sell them.  My latest
working  ErgoTravel is the choc version and being much thinner (and ble
too) it's an ideal travel keyboard.  Email me at jpconstantineau@gmail.
It's on the list of extras I have post GB.

themmes, 2019-04-04:
@jpconstantineau are these the one you dubbed ErgoTravelLight on the Githubpage?

jpconstantineau, 2019-04-04:
Yes. I do have some on hand if you are interested.

themmes, 2019-04-04:
With some, you mean PCBs? Very cool! Would these also work with the new Kailh X type switches?

jpconstantineau, 2019-04-04:
Humm  Good question about those new switches.  The footprint is not the
same and the plate cutout is similar.

Both switches uses a 14mmx14mm switch plate cutout.  The locations of the
pins is different.
The choc switch has a hole for a LED.  The X doesnt.

By comparing the drawings here:
http://www.kailh.com/en/Products/Ks/NotebookS/649.html
and here: http://www.kailh.com/en/Products/Ks/CS/319.html
Looks like the PCB would need to be updated to support those.

Getting the switches and compatible keycaps would be the harder part....

themmes, 2019-04-04:
I thought I read somewhere that the X has LED built in? Well, I am not too interested in backlight for now anyways..

I think the X Switch can be ordered here; https://nl.aliexpress.com/store/product/Kaih-Low-profile-Notebook-X-Switch-keyboard-switch-for-laptop-tactil-type-Pretravel-1-2mm-clikcy/4670072_32985602923.html?spm=a2g1y.12024536.productList_12815140.pic_7

Hmm.. I am not familiar with drawing PCBs but I could give it a try. I saw you were using KiCAD software? Are there any other parts I should adapt that you know of?

jpconstantineau, 2019-04-04:
looking at the picture of the switch itself on aliexpress, there is a
cutout at the bottom of the switch for a LED.

They are not cheap at 1.30USD/switch...  Keycaps available from Kailh are a
bit boring... blank whites...
For an ErgoTravelThin (?) you would need 6 packs of 10 switches and 2 packs
of 50 keycaps.  For a total of 107USD. That's almost twice the price of
choc switches...

You would need to use Kicad:
1 - create the switch footprint in the footprint editor
2 - replace the mapping between schematic and PCB for the switches to point
to new footprint
3 - export NET
4 - Go to PCB editor
5 - import NET, replace footprints
6 - re-route all wires
7 - run design check/pour copper planes
8 - create Gerber
9 - send gerber to PCB fab.

Perhaps someone already has the X switches footprint - done in a reversible
way (we need it to be reversible since we flip the PCB upside down for the
left/right halves)

themmes, 2019-04-04:
Thanks for this extensive overview! I agree that these are very expensive switches.

If using the Kailh Choc switches, what do you have laying around that I could buy? PCB or PCB with casing or full kit?

jpconstantineau, 2019-04-04:
Email me at jpconstantineau@gmail and we can continue offline... I have
full kits

closed 2021-11-24 by jpconstantineau