jpconstantineau/ErgoTravel #32: Question: Partially assembled ErgoTravel ProMicro attachment

berfarah, 2018-12-11:
Hiya! Please let me know if you prefer that questions be in a thread on reddit, I'm happy to oblige and move this to wherever makes sense.

I opted for the partially built keyboard, which means that the diodes and Pro Micro headers were already on the board (thank you).

Looking at the guide, it looks like you solder diode pins to the pro-micro rather than the header pins I got. I got header pins in my kit and I just want to verify before I goof up my build:
- Are header pins intended to be soldered onto the pro micro / do they fit into the pro-micro headers?
- Should I get and use diode legs instead?

I tested individual header pins and it doesn't seem like they stick in the sockets, though I could just not be trying hard enough.

jpconstantineau, 2018-12-11:
Headers won't fit in the sockets.

For wires, you can use diode legs or any wire that's thin enough to fit
securely in the sockets.

24awg or 26awg wire should work well with the sockets. Anything bigger and
you will have trouble removing the pro micro. Anything thinner and you may
have intermittent connections.

The two types of sockets I have used have slightly different diameters and
the 24awg is on the larger side for the low profile sockets but work well
for the tall sockets.

berfarah, 2018-12-11:
@jpconstantineau awesome, thanks! That answers the question, I'll close this issue.

closed 2018-12-11 by berfarah