jpconstantineau/ErgoTravel #33: Acrylic Case Path Lengths and Thickness

sedgwickc, 2019-04-05:
I'm looking to have an acrylic/steel combo case cut for an ErgoTravel build but I have some questions about the case files in the repo:

1. Are the files in acryliccase/jpconstantineau/ compatible with the middle section defined in the grogdor folder? 

2. What are the dimensions and path lengths for the top1u1u, middle and bottom plates defined in the grogdor folder (or is there some way to determine these values from the svg files provided)? 

3. What thickness of acrylic should be used when getting the middle section cut (assuming the top and bottom are both acrylic)?

jpconstantineau, 2019-04-05:
1 - files in  acryliccase/jpconstantineau/ should not be relied upon.  In
fact, they should be removed since they have not been used for production
2 - I don't know about this. Gorgdor did all the work on the laser cutter
he has access at his maker space.
3 - we used a stack of 4 3mm sheets for the middle sections.  the same 3mm
as the top/bottom plates.

sedgwickc, 2019-04-05:
Thanks for the response @jpconstantineau! So it sounds like I should be using @Grogdor's version of the case files then. I was planning to go with a thinner top plate since the stainless steel will be strong enough at the same thickness as the PCB plates to serve its purpose so I'm glad that you mentioned that. 

@Grogdor: are you able to provide the path length and dimensions for the 1u1u top plate, middle sections and bottom plate (or the .dxf files that contain information)?

Grogdor, 2019-04-07:
Hello,

1) I'll submit a PR to sync the production files from my repo, but be warned they're a mess. There's the regular wired version, another for Bluemicro, remnants of "tented" designs which were never used, and various DFM iterations.
2) Use my DXFs with your workflow. They're metric and generally sane, may want to "overkill" them and verify polylines, dimensions.
3) Check out the pics/ for stackup, a single 12mm layer of acrylic would be nicest/strongest. If I recall, you can get away with 9mm for low-profile headerless Arduino mounting, and possibly Bluemicro. We used multiple 3mm layers for stackup flexibility, ease of manufacturing and cost.

For your metal top plate, use switch spec of 1.5mm plate thickness and remove the engrave (black box) near the 2u thumb switch. Beware I found some extra red cut lines inserted around the engrave when importing/printing this to the laser cutter, so verify the measurements because dxf fills seem to get mangled. Are you cutting yourself, or sending this off? Appreciate any feedback, and pics of the result.

sedgwickc, 2019-04-15:
Thanks @Grogdor for the PR, your files are exactly what I needed! I have made a version of the "top only bluemicro DFM9.dxf" file using LibreCAD that is missing the stab engrave as well as cutlines and the measurements still look sane. I submitted it to LaserBoost for cutting from stainless steel at 1.5mm. I'll update with the result and pictures once the plates arrive.

jpconstantineau, 2019-04-15:
Sounds great!

Once you get your plates and validate that it's all good, you could add
your cad files to the repo. This way, the next person who would want metal
plates, would be able to start from your files.

Looking forward to pictures...

sedgwickc, 2019-05-19:
The plates have arrived! I don't have any bottom plates yet so I can't confirm that the top plate holes will match up but the switch holes line up nicely so I'm not too worried. I'll clean up my local repo then submit a pull request soon with the pictures and design files. Here are some pictures with the plate and PCB held together with some switches which aren't soldered in place:

[image]

[image]

jpconstantineau, 2019-05-19:
These look great. I'll approve the PR when I get it.