jpconstantineau/ErgoTravel #36: Some questions about ErgoTravelLight 3.01

Ardakilic, 2022-04-30:
Hello, 

First of all, thanks for making this board available for us all! I loved the Cherry MX variant and printed a choc variant, but I have some questions. 

I printed ErgoTravelLight v3.01 by/for myself without even checking #29 (I built the cherry version back then and I wanted to make a choc variant), while ordering other stuff from JLCPCB, and I'm currently a bit confused:

Is this version even checked and confirmed working? I received the boards this week from JLC and when I check now I'm a bit confused. Are there any guides available ?

1. Are the orientation of Pro micros same with Cherry variant, as in build guide?
2. What is the diodes' orientation? I cannot discriminate it on the PCB silk screen. Any photos available to check ? Or, Is the diodes' orientation same as the Cherry MX guide, where the zener diodes' black lines are in same orientation as SMD ones' vertical lines?
3. Which LEDs would you prefer for soldering ? I have ws2812b 3535, but I believe it'd be impossible to solder these with iron, so cream flux + heat would be required since the foot print is too small.

Thanks in advance!

jpconstantineau, 2022-04-30:
RGB LEDs: SK6812-EC1515
These are 1.5mm x 1.5mm to be able to fit in the hole of the switch.  They are very small and quite difficult to install. As you indicated, there is no clear indication of the polarity on how to install them. LEDs are installed on the same side as thee switches.

Zener diodes: There aren't any zeners. They should be standard 1N4148 in the surface mount type - do not use through-hole diodes as these will cause problems with the top plate.  As for their orientation, I believe there is a larger band but would have to double check on the actual PCB.  I do have a set assembled here, I could take pictures and attach here. Diodes should be installed the other side from the switches.

Pro Micros: most robust way to check the orientation is with a multimeter. The pro micro has 2 GND pins next to each other on one side and this should match what's on the board.  They get installed on the same side as the diodes, opposite side to the switches.  Remember to use sockets or make sure not to solder them in the wrong sequence (or you won't be able to install  the switches)

There aren't any build guides for this one and I only build a set.  It's best to get the circuit diagram printed as there are a few components that might need a bit of clarity on where they go, how they are connected and what values are needed.

Ardakilic, 2022-05-01:
Hello again,

Regarding the diodes; I was mentioning about the polarity of them. Since I believe the orientation is same with the cherry one's zener diodes, which are shown in the build guide pictures, I was asking whether the polarity and orientation is same with the one on the guide. I'll use 1n4148 sod-123 diodes. If the position of the black lines on the zener diodes, which is shown on the build guide, match with the lines of the SMD diode chips, it's okay. If not, I'd love to take a look at them, if it wouldn't be a bother to you. Or at least the photo of the PCB/screenshot of the PCB design with silk screen that indicates the diodes' positions would be great.

Regarding the pro micros. Thanks! I'll use sockets, and just did not want to burn them if I made something stupid, since they are becoming way more expensive due to chip shortage.

Thanks! I'll also try to check with the circuit diagram if I'm stuck. I hope everything goes well.

jpconstantineau, 2022-05-01:
I'll get a few pictures.

The diodes for the regular ergotravel, they are just small switching diodes of the type 1n4148 package DO35.  Zener diodes are used for something completely different.  They have a specifically designed reverse voltage at which they will conduct in reverse.  They are used as voltage references or in low-efficiency voltage regulation applications.  Zener diodes are not used for keyboards.  The Ergotravel build pictures show a glass DO35 package small signal diode, not a zener diode.  They might look the same, however, their behavior is quite different.

My standard surface mount diodes are of SOD-123 package.  I'll get a few pictures included here later tonight.

jpconstantineau, 2022-05-01:
Top View - Board Boards
[image]

Bottom View - Both boards
[image]

Surface Mount Diode. Note position of Bar on Diode lines up with bounding box/line on silkscreen
[image]

Position of RGB LED (Note black dot at top right)
Also note orientation of RAW pin of Pro Micro
[image]

Position of RGB LED (Note absence of black dot at top right) - other half
Also note orientation of RAW pin of Pro Micro
[image]

Note that for the RGB LEDs to be powered on and to receive their data, a few 0-ohm resistors need  to be installed.  See below:
[image]

[image]

Ardakilic, 2022-05-01:
Hello again, 

Thanks for these. Seriously! This will help a lot! 🙏 

Here's what JLCPCB sent me:

[image]
[image]

I believe JLC messed up my silk screens. I don't have these indicator line between two other lines on diodes.

What value are these caps btw? Would 100nf work ?

Also, can you confirm please? If I skip the LEDs, I won't need 0ohm resistors and the caps, correct ?

Thanks again for the effort!

Ardakilic, 2022-05-03:
I have soldered diodes, TRRS connector and pro micro sockets. When pro micros and switches arrive I'll try to test whether it's working as expected or not.

Ardakilic, 2022-08-01:
Hey @jpconstantineau , apologies foe the huge delay. I changed my job, and renovated my house, so the delay happened.

Here's what I achieved so far:

[image]

[image]

This is currently working as a split keyboard with a TRRS cable. I have just not decided how should I fit the bottom.

I could not use top plate, because the pro micro connector/socket's solder joints' height makes the top plate not fit between board and switch. It needs to be thinner than 1.6mm. The worst part is JLCPCB asked for $15 more because it had "too many holes", apart from the $9.3 initial price. The thin edges may be the reason for this.

I could not solder these RGB LEDs, so that's why I did not solder any caps. I even fail mostly at while soldering WS2812B 5050 LEDs. The smallest component that I can solder by hand is 0603 components. I say these LEDs are almost impossible to solder by hand, at least for me.

I'll try LED strips in the future, I had tried that on Cherry MX version last year and it worked great.

Thanks again for making this available!

Ardakilic, 2022-08-04:
I'm closing this issue because there's nothing left of me for to ask. As mentioned, maybe the choc version's plate can be cheaper, apart from that everything works as expected. Thanks once again!

closed 2022-08-04 by Ardakilic

jpconstantineau, 2022-10-11:
I see, JLC des have the tendency to edit gerbers before they send them to
production.
If you load the gerber on this site, you should be able to see all the
details.
https://tracespace.io/view/

Without RGB LEDs, you don't need these 0 ohm resistors or the capacitors.